knaufinator/6DOF-Rotary-Stewart-Motion-Simulator #10: Wrong PCB hole diameter

krotos139, 2020-11-19:
Wrong PCB holes for screw terminals.

knaufinator, 2020-11-19:
to small?  I had these blue ones laying around in my shop, and they fit. suggestions?
[image]

krotos139, 2020-11-19:
I bought this screw terminals:
[image]
This terminals looks like your. This screw terminal has pin thikness 1mm:
[image]
But PCB has hole size is 0.915 mm. And I cannot insert it easy.

Have you had any problems installing these terminals?

knaufinator, 2020-11-19:
Mine were used/desoldered from an old project, they fit.. but it was snug,.. I assumed because of the solder-remains 

I checked the part in the schematic and it looks like @0.915mm diam. so spot on measure..

I have found based on your picture, this schematic part that looks like it would fit better, 
[image]

I will update the PCB's with this screw terminal, I agree that this would be better. thanks... I will try to do it today

knaufinator, 2020-11-19:
Updated both PCB Gerbers, with 1.1mm pitch screw terminals, Thanks @krotos139

closed 2020-11-19 by knaufinator