knaufinator/6DOF-Rotary-Stewart-Motion-Simulator #16: simtools is free open source？i cant find it website

zouzhe1, 2022-11-03:
Also, please tell me what the following three libraries are and where to download them. You didn't make it clear
**Bounce2.h
BLE2902.h
DigitalFilters.h**

knaufinator, 2022-11-03:
https://simtools.us/

bounce2 and ble9202 are 3rd part libraries that should be/... at least were available through arduino library manager,..

I could see about packaging up the versions I have now and placing them in the project so it is self contained,... I will look at doing that this weekend...

Digitalfilter.h... is in the same directory, as all the other source files, .. but its not used so could be scrapped out if ya want.

knaufinator, 2022-11-09:
updated Arduino controller project to include BLE and bouce2 libraries in the package

zouzhe1, 2022-11-10:
@knaufinator 
thanks。i want to make it。
I have a few questions for you.

Is there any introduction to the calculation in your helpers.cpp?

Do you use absolute value coded motors?

Does the motor have a brake?

knaufinator, 2022-11-10:
take a read at - https://www.xsimulator.net/community/threads/compact-ac-servo-6dof-rotary-stewart-platform.14769/

This is the forum/page that I and others post to for this project.

It will offer up some details that of which you seek.

There is an excel document in the Technical folder in this GitHub project, this contains the reverse kinematic algorithm, as well allows you to change parameters and see the result in a pseudo 3d visualization.... 

I took that algorithm and converted it to c++, and then built allot of other code around that to push my servos around.

understanding that excel algorithm is key in order to make dimensional changes to the layout of the motors. it needs to be opened from actual excel.... not a open source clone like open office its an old file at this point,.. but should still work in the latest version of MS Excel. anything you can get the excel sheet to do with parameter changes, you can do in my  controller.
i.e. motor orientations, arm lengths, chassis size...so on.

I don't look at the encoding of the motors, the controller keeps track of where it is based on position ~-4000 -> 4000 or so positions with 0 being the home middle position.

The motors do not have a brake, when power is released the rig drops gracefully to the ground. 

position is recorded at power up, the ac servos will home themselves based on a magnetic sensor at the top dead center 12 o'clock position, then rotate back to middle/home, based on a calibrated step count.