knaufinator/6DOF-Rotary-Stewart-Motion-Simulator #19: Adjust position pulse output speed (frequency)

1248280302, 2023-04-25:
How is the speed(frequency) of the position pulse output controlled? I tried to modify the parameter ‘maxRate’, but it didn't work.
#define MICRO_INTERVAL_FAST  10
How is this value calculated?

knaufinator, 2023-04-25:
when not in debug mode, the microInterval = MICRO_INTERVAL_FAST = 10(microseconds)

 i.e. line 34 in controler.ino- int microInterval = MICRO_INTERVAL_FAST;

which is time at a minimum,... the code just checks to make sure that at least 10 microseconds has passed before allowing the next iteration to change.

i.e. line #485 in controll.ino
 //creates the pulses in realtime, if enough time has passed.
  if (dif >= microInterval) {
          
if you changed this microInterval to 100 microseconds, as it does when in debug mode, then each pulse interval waits at least 100 microseconds after the last. and this definitely slows down the output. 

if you are trying to make it less than 10 microseconds... you would be running into speed bottlenecking limitations in the ESP32 ->GPIO  SPI communications (maxed out at 10MHZ already), I likely came to the conclusion at the development time that 10 microseconds was about the fastest... it very well could be 25 microseconds are passing,.. but I just make sure that at least 10 has. You could time it and spit it out over serial, just don't print from the same thread/core... as serial is very time consuming.

the maxRate i.e. "float maxRate = .001;" parameter is just used to slow down just after resetting the estop,.. so you don't slam to the requested position,... instead you smoothly glide up to it. 

hope that clarifies it, if not elaborate more, and I will see what I can do.

1248280302, 2023-04-29:
@knaufinator Well, in the reality that PWM cannot reach 10MHz frequency and the rated motor speed is 3000rpm, how should the relevant pulse speed limit be modified? I can't find the relevant function.

knaufinator, 2023-04-29:
The SPI communication speed is 10 MHz, which is not the same as the achievable pulse output speed. It determines how quickly we can are talking to and sending the next set of output iterations to the GPIO splitter. Additionally, the signal being sent is not PWM but rather individual pulses, each with the same width defined by '#define MICRO_INTERVAL_FAST 10', with a minimum time delay of 10 microseconds.

1248280302, 2023-04-29:
@knaufinator Thanks!If I set output parameter “interval loops = 1 (2ms)”, ESP32 seems to have a problem receiving data for some axes. Is this normal?If I set “interval loops > 3” and it will running normally.

knaufinator, 2023-04-29:
hard to say, its been so long since I tested this. may be a signal reliability issue, maybe a few of your AC drivers are a little more tolerant to the shorter signal. The next phase I never got around to, would be to use a differential signal, that provides complimentary signals for step and direction,... this could then be used to increase the reliable pulse speed. but would require a new PCB design... for hobby purposes, the PCB was fine as is... you just need to add virtual gearing to the AC driver in order to make up for the loss of precision due to reduced steps for the desired top speed.

closed 2025-02-09 by knaufinator