knaufinator/6DOF-Rotary-Stewart-Motion-Simulator #21: License terms

wotupfoo, 2024-10-27:
Hi,
I want to fork your project but I see in your controller code a copy of the RotaryStewartPlatorm code - like getAlpha(), setPos() without the license attribution anywhere I can find in your repo. 
Is this a derivative of that work, or did you write them and the other repo is a copy of yours (and there they didn't follow the license terms). 

Just trying to do the right thing as I move off of his code base to use some of yours instead since I like your SimHub|SimTool integration more. 

https://github.com/ThomasKNR/RotaryStewartPlatform/blob/master/src_arduino_code/platform.ino

knaufinator, 2024-10-27:
yeah, do what ever ya want with the code my dude. If you do something cool, send a link my way to take a look at it.  have fun

ChristianKnorr, 2024-10-27:
Why do you want to fork? And yes, please link it here 👍

wotupfoo, 2024-12-04:
I'd rather not fork but my playing around has already deviated a lot from what's in the repo here.
It's always a debate about whether it should be a completely different code base in a branch or a fork.

Here it is so far, the only code difference is a bit of code cleanup to suite my liking.
The big "value add" right now is that I captured all my research into README.md markdown files.
For example, here is my research on Mixed Reality gaming (my goal is to have a Stewart platform with Mixed Reality).

https://github.com/wotupfoo/SimRacing/blob/main/webpages/Mixed%20Reality%20Cockpit/README_OpenXR.md

closed 2025-02-09 by knaufinator